eclipse-zenoh/zenoh-pico #871: [Bug] z_get() callback not called in async mode, z_recv() stuck in sync mode

pdematteis, 2025-01-30:
### Describe the bug

When a query reply is received, `_z_trigger_query_reply_partial_inner()` is called to deal with it.
It locks the session mutex, then call `_z_reply_copy()` to store the reply out of the context. This function fails and, since it is called by the _Z_RETURN_IF_ERR macro, it leaves the mutex locked.
`_z_reply_copy()` fails because it calls `_z_sample_copy()`, which calls `_z_bytes_copy()` in turn and this latter does not handle zero-size slices in the right way, returning _Z_ERR_SYSTEM_OUT_OF_MEMORY when z_malloc(0) is called.
For me, it happens that attachment is a zero-size slice, but also the payload may be empty (or even NULL?).

### To reproduce

.

### System info

.

pdematteis, 2025-01-30:
In `_z_trigger_query_reply_partial_inner()`, I think that

`_Z_RETURN_IF_ERR (_z_reply_copy(&pen_rep->_reply, &reply));
`

should be replaced with

`_Z_CLEAN_RETURN_IF_ERR(_z_reply_copy(&pen_rep->_reply, &reply), _z_session_mutex_unlock(zn));
`

sashacmc, 2025-01-30:
Hello @pdematteis, thanks!

closed 2025-01-30 by milyin

pdematteis, 2025-02-03:
A fix for FreeRTOS is:

```
#define BADPTR ((void *)(~0))

void *z_malloc(size_t size) { return size > 0 ? pvPortMalloc(size) : BADPTR; }

void z_free(void *ptr) { if (ptr != BADPTR) vPortFree(ptr); }
```