eclipse-zenoh/zenoh-pico #884: [Bug] Mapped key-expressions miss the trailing / in version 1.2.0

pdematteis, 2025-02-04:
### Describe the bug

I'm using a zenoh-pico 1.2.0 subscriber that subscribe `ft/**`.
A zenoh-c publisher sends topics like `ft/master/led/1`.
- using zenoh-c binaries 1.1.1, `/master/led/1` is sent over the wires and received by the subscriber
- using zenoh-c binaries 1.2.0, `master/led/1` is sent instead
The subscriber reconstructs the keyexpr by merging the received part with the subscribed portion, that is `ft`.
With version 1.2.0 publisher, this results in a wrong `ftmaster/led/1` key, that is dropped, while in version 1.1.1 it works with no issue.

This is probably NOT a Zenoh-pico issue but a zenoh-c api one. Not sure about that and not easy to verify from my side.

Thanks.

### To reproduce

.

### System info

.

jean-roland, 2025-02-04:
I don't know if it's related to your issue but keyexpression starting with a `/` are invalid. I don't remember if it's specified in our [keyexpr RFC](https://github.com/eclipse-zenoh/roadmap/blob/main/rfcs/ALL/Key%20Expressions.md), but it should.

pdematteis, 2025-02-04:
It is not starting with /, it is like `ft/master/led/1`.
The subscriber registers `ft/**`, that is internally recorded like just `ft` with an ID.
For this the publisher seams to send only the extra portion of the string with the matching ID, that is `/master/led/1`, but v1.2.0 drops the leading / while v1.1.1 does not.
I don't know what exactly is expected under the hood, but something looks broken in the newer version.

jean-roland, 2025-02-04:
Ok then I just tested and it works on pico so I suggest you reopen this issue on the Zenoh-C repo.

closed 2025-02-04 by jean-roland

pdematteis, 2025-02-04:
Yes pico sub vs pico pub does work for me as well.
Will check it better in zenoh-c.
Thanks

pdematteis, 2025-02-04:
FYI: https://github.com/eclipse-zenoh/zenoh-c/issues/904