eclipse-zenoh/zenoh-pico #887: zenoh pico on openwrt (teltonika RUT240)

rammfr, 2025-02-05:
### Describe the feature

I try to run zenoh-pico on the Teltonika device RUT240 (Arch: mips)
Using the SDK, I am able to compile zenoh-pico v1.2.0 and install it on the target hardware.

Network configuration seems to be ok:

root@RUT240:~/zenohpico_examples# ping 192.168.1.99
PING 192.168.1.99 (192.168.1.99): 56 data bytes
64 bytes from 192.168.1.99: seq=0 ttl=64 time=0.699 ms
64 bytes from 192.168.1.99: seq=1 ttl=64 time=0.681 ms
...

zenohd is running on my dev machine:
...
INFO main ThreadId(01) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.1.99:7447
...

Trying to run the examples I got following:

root@RUT240:~/zenohpico_examples# ./z_info 
Opening session...
Unable to open session!

root@RUT240:~/zenohpico_examples# ./z_scout 
Scouting...
Did not find any zenoh process.

How do I debug this issue?

sashacmc, 2025-02-05:
Hello @rammfr ,

when running examples without parameters, the multicast scouting mode is used.
Perhaps you have the corresponding ports blocked or the routes not configured.

Try to explicitly specify the connection: -e tcp/192.168.1.99:7447

For debug an issue you can run Zenoh with trace log enabled (with env variable: RUST_LOG=trace)
and compile Zenoh-Pico with logs enabled (with cmake parameters `-DCMAKE_BUILD_TYPE=Debug` `-DZENOH_DEBUG=3`)

rammfr, 2025-02-07:
Hello @sashacmc,
thanks, that helps! Now, zenoh-pico on the embedded device finds now the router:

root@RUT240:~/zenohpico_examples# ./z_info -e tcp/192.168.1.99:7447
Opening session...
Own ID: 3E9623FB43D2B43818946E5324F023EA
Routers IDs:
 C7F546160666BA639872168DA055544E
Peers IDs:

and is able to communicate with my dev machine:

root@RUT240:~/zenohpico_examples# ./z_pub -e tcp/192.168.1.99:7447
Opening session...
Declaring publisher for 'demo/example/zenoh-pico-pub'...
Press CTRL-C to quit...
Putting Data ('demo/example/zenoh-pico-pub': '[   0] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   1] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   2] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   3] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   4] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   5] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   6] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   7] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   8] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[   9] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  10] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  11] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  12] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  13] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  14] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  15] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  16] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  17] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  18] Pub from Pico!')...
Putting Data ('demo/example/zenoh-pico-pub': '[  19] Pub from Pico!')...
Segmentation fault

% ./z_sub 
Opening session...
Declaring Subscriber on 'demo/example/**'...
Press CTRL-C to quit...
>> [Subscriber] Received PUT ('demo/example/zenoh-pico-pub': '[   3] Pub from Pico!')
>> [Subscriber] Received PUT ('demo/example/zenoh-pico-pub': '[   4] Pub from Pico!')
>> [Subscriber] Received PUT ('demo/example/zenoh-pico-pub': '[   5] Pub from Pico!')
...

Unfortunately, after a while (not reproducible) a segmentation fault occurs.  
I guess I need to fine tune the buffer sizes, etc.
Can you give me some hints on that?

Furthermore, subscribing on the embedded device does not

root@RUT240:~/zenohpico_examples# ./z_sub -e tcp/192.168.1.99:7447
Opening session...
Declaring Subscriber on 'demo/example/**'...
Press CTRL-C to quit...

when sending on the dev machine 

 % ./z_pub
Opening session...
Declaring Publisher on 'demo/example/zenoh-rs-pub'...
Press CTRL-C to quit...
Putting Data ('demo/example/zenoh-rs-pub': '[   0] Pub from Rust!')...
Putting Data ('demo/example/zenoh-rs-pub': '[   1] Pub from Rust!')...
Putting Data ('demo/example/zenoh-rs-pub': '[   2] Pub from Rust!')...
Putting Data ('demo/example/zenoh-rs-pub': '[   3] Pub from Rust!')...

BR Franz

sashacmc, 2025-02-07:
Hello @rammfr,

could you please provide logs from Zenoh (for simplicity you can just run rust examples without router) and Zenoh-Pico.
And stack trace for "Segmentation fault" (you can use gdb for example) if it is possible.