eclipse-zenoh/zenoh-pico #897: [Bug] UDP transport destabilizes above 1500 bytes

vortex314, 2025-02-18:
### Describe the bug

I'm working with an ESP32 and the default protocol it communicates with the broker is UDP.
When I increase the message size gradually it looks to become unstable above 1500 bytes .

### To reproduce

Running a router locally as docker container 
```
docker run --init --name=zenoh --net=host -i -e "RUST_LOG=debug" -v $(pwd):/zenoh -d --restart=always eclipse/zenoh -c /zenoh/config.json5 

```
The ESP32 is acting as zenoh client , works correct until message size above 1500 bytes 
```
assert failed: spinlock_acquire spinlock.h:140 (lock->count == 0)

Backtrace: 0x4008283a:0x3ffcde80 0x400894cd:0x3ffcdea0 0x400912c1:0x3ffcdec0 0x4008a566:0x3ffcdfe0 0x4008a20d:0x3ffce010 0x400e0c10:0x3ffce030 0x4008a2e1:0x3ffce050
  #0  0x4008283a in panic_abort at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/esp_system/panic.c:463
  #1  0x400894cd in esp_system_abort at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/esp_system/port/esp_system_chip.c:92
  #2  0x400912c1 in __assert_func at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/newlib/assert.c:80
  #3  0x4008a566 in spinlock_acquire at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/esp_hw_support/include/spinlock.h:140 (discriminator 1)
      (inlined by) xPortEnterCriticalTimeout at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:474 (discriminator 1)
  #4  0x4008a20d in vPortEnterCritical at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:567
      (inlined by) xEventGroupSetBits at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/freertos/FreeRTOS-Kernel/event_groups.c:566
  #5  0x400e0c10 in z_task_wrapper at .pio/libdeps/esp1/zenoh-pico/src/system/espidf/system.c:62
  #6  0x4008a2e1 in vPortTaskWrapper at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

```

Start zenoh rust z_pub
```
./z_pub -m client -k dst/esp1/ota -p "aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaazzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzaaaccccccccccccqqqqqqqqqqqqqq"
```

### System info

- Router runs on a linux system as docker container

vortex314, 2025-02-18:
I suspect the UDP fragmentation happening at that limit 

[image]

vortex314, 2025-02-18:
Smells like this issue : https://github.com/docker/for-win/issues/1144
Docker seems to drop UDP messages above the MTU

jean-roland, 2025-02-19:
Yeah that looks like network packet fragmentation related issue, tcp abstract the mtu so it explains why you observe this only on udp. Tell me if using a docker directly on the system solves the problem, otherwise we'll have to investigate

vortex314, 2025-02-19:
I tried outside a docker container the effect is the same, so it doesn't seem docker related. 
Testing with TCP I can go to 4K before it fails . Failing in the sense that I see on Wireshark the data is send to the ESP32 , but is silently dropped .

vortex314, 2025-02-19:
Testing with docker and UDP again, shows that the issue is solved 
`

[image]

This was a very deep dive. ;-)

closed 2025-02-19 by vortex314