eclipse-zenoh/zenoh-pico #898: [Bug] Setting encoding to CBOR creates a crash

vortex314, 2025-02-19:
### Describe the bug

Setting the put_options for encoding to cbor, it creates a crash in the lower level layers

### To reproduce

If the encoding is left to the default , send and receive works fine
```c++
{
  _timer_publish = timer_repetitive(1000); // timer for publishing properties
  z_put_options_default(&put_options);
  z_owned_encoding_t encoding;
  z_result_t r = z_encoding_from_str(&encoding, "application/cbor"); // ZP_ENCODING_APPLICATION_CBOR;
  if (r != Z_OK)
  {
    INFO("Failed to get encoding for application/cbor ");
  }
  put_options.encoding = z_move(encoding); // commenting this line out makes it work fine
  put_options.congestion_control = Z_CONGESTION_CONTROL_DROP;
  prefix("device"); // default prefix
}
```
```c++
{
  _timer_publish = timer_repetitive(1000); // timer for publishing properties
  z_put_options_default(&put_options);
  z_owned_encoding_t encoding;
  z_result_t r = z_encoding_from_str(&encoding, "application/cbor"); // ZP_ENCODING_APPLICATION_CBOR;
  if (r != Z_OK)
  {
    INFO("Failed to get encoding for application/cbor ");
  }
  put_options.encoding = z_move(encoding); // commenting this line out makes it work fine
  put_options.congestion_control = Z_CONGESTION_CONTROL_DROP;
  prefix("device"); // default prefix
}
```
Crash dump backstrace on ESP32 
```
[I (2364) esp_netif_handlers: sta ip: 192.168.0.247, mask: 255.255.255.0, gw: 192.168.0.1
I 00:00:01.820 |  wifi_actor.cpp: 160 | WiFi STA got IP address
I 00:00:01.856 | zenoh_actor.cpp: 148 | Own ZID  : 69652b3eda3fd1b6a08eede46cba79b5
I 00:00:01.857 | zenoh_actor.cpp: 155 | Connected to Router : b2884c05063d1566c1ca1da6cf0050ce
I 00:00:01.862 | zenoh_actor.cpp:  75 | Connected to Zenoh.
I 00:00:01.867 | zenoh_actor.cpp: 213 | Declaring subscriber for 'dst/esp1/**'...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c29b  PS      : 0x00060e30  A0      : 0x800e69c8  A1      : 0x3ffbf050  
A2      : 0x3ffcf7cd  A3      : 0x00000001  A4      : 0x00000067  A5      : 0x3ffcf7cd  
A6      : 0x00000000  A7      : 0x3ffbf180  A8      : 0x800e6d36  A9      : 0x3ffbf020  
A10     : 0x3ffcf79c  A11     : 0x00000003  A12     : 0x00000001  A13     : 0x00000005  
A14     : 0x00000000  A15     : 0x3ffcaba0  SAR     : 0x00000007  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x4000c298:0x3ffbf050 0x400e69c5:0x3ffbf060 0x400e6d75:0x3ffbf080 0x400e828a:0x3ffbf0b0 0x400e8346:0x3ffbf0d0 0x400e4752:0x3ffbf0f0 0x400e518e:0x3ffbf130 0x400e58f1:0x3ffbf160 0x400e132f:0x3ffbf180 0x400e14b1:0x3ffbf1c0 0x400e14f8:0x3ffbf260 0x400dd729:0x3ffbf280 0x400dc37b:0x3ffbf470 0x400db3e5:0x3ffbf520 0x400dbcb9:0x3ffbf5a0 0x400dbeee:0x3ffbf680 0x400d2bc9:0x3ffbf6b0 0x400d2c35:0x3ffbf6d0 0x4008a2e1:0x3ffbf6f0
  #0  0x400e69c5 in _z_iosli_write_bytes at .pio/libdeps/esp1/zenoh-pico/src/protocol/iobuf.c:108 (discriminator 1)
  #1  0x400e6d75 in _z_wbuf_write_bytes at .pio/libdeps/esp1/zenoh-pico/src/protocol/iobuf.c:413
  #2  0x400e828a in _z_string_encode at .pio/libdeps/esp1/zenoh-pico/src/protocol/codec.c:364 (discriminator 2)
      (inlined by) _z_string_encode at .pio/libdeps/esp1/zenoh-pico/src/protocol/codec.c:361 (discriminator 2)
  #3  0x400e8346 in _z_encoding_encode at .pio/libdeps/esp1/zenoh-pico/src/protocol/codec.c:401
  #4  0x400e4752 in _z_push_body_encode at .pio/libdeps/esp1/zenoh-pico/src/protocol/codec/message.c:286
  #5  0x400e518e in _z_push_encode at .pio/libdeps/esp1/zenoh-pico/src/protocol/codec/network.c:65
  #6  0x400e58f1 in _z_network_message_encode at .pio/libdeps/esp1/zenoh-pico/src/protocol/codec/network.c:498
  #7  0x400e132f in _z_transport_tx_send_fragment_inner at .pio/libdeps/esp1/zenoh-pico/src/transport/common/tx.c:62
      (inlined by) _z_transport_tx_send_fragment at .pio/libdeps/esp1/zenoh-pico/src/transport/common/tx.c:90
  #8  0x400e14b1 in _z_transport_tx_batch_overflow at .pio/libdeps/esp1/zenoh-pico/src/transport/common/tx.c:160
      (inlined by) _z_transport_tx_send_n_msg_inner at .pio/libdeps/esp1/zenoh-pico/src/transport/common/tx.c:218
      (inlined by) _z_transport_tx_send_n_msg at .pio/libdeps/esp1/zenoh-pico/src/transport/common/tx.c:258
      (inlined by) _z_transport_tx_send_n_msg at .pio/libdeps/esp1/zenoh-pico/src/transport/common/tx.c:246
  #9  0x400e14f8 in _z_send_n_msg at .pio/libdeps/esp1/zenoh-pico/src/transport/common/tx.c:434
  #10 0x400dd729 in _z_write at .pio/libdeps/esp1/zenoh-pico/src/net/primitives.c:244
  #11 0x400dc37b in z_put at .pio/libdeps/esp1/zenoh-pico/src/api/api.c:897 (discriminator 4)
  #12 0x400db3e5 in ZenohActor::zenoh_publish(char const*, std::vector<unsigned char, std::allocator<unsigned char> > const&) at src/zenoh_actor.cpp:313 (discriminator 3)
  #13 0x400dbcb9 in ZenohActor::on_cmd(ZenohCmd&) at src/zenoh_actor.cpp:104 (discriminator 3)
  #14 0x400dbeee in Actor<ZenohEvent, ZenohCmd>::handle_all_cmd() at src/actor.h:176
  #15 0x400d2bc9 in Thread::run() at src/actor.cpp:267
  #16 0x400d2c35 in Thread::start()::{lambda(void*)#1}::_FUN(void*) at src/actor.cpp:181
      (inlined by) _FUN at src/actor.cpp:181
  #17 0x4008a2e1 in vPortTaskWrapper at /home/lieven/.platformio/packages/framework-espidf@3.50301.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 7f1d7e6cb

Rebooting...
ets Jun  8 2016 00:22:57
```

### System info

- ESP32 - ESPIDF 5.31.
```

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080638
I (31) boot: ESP-IDF 5.3.1 2nd stage bootloader
I (31) boot: compile time Feb 18 2025 20:47:53
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (76) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (84) boot:  2 app0             OTA app          00 10 00010000 00170000
I (91) boot:  3 coredump         Unknown data     01 03 00180000 00020000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2314ch (143692) map
I (161) esp_image: segment 1: paddr=00033174 vaddr=3ffb0000 size=0400ch ( 16396) load
I (167) esp_image: segment 2: paddr=00037188 vaddr=40080000 size=08e90h ( 36496) load
I (182) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a242ch (664620) map
I (410) esp_image: segment 4: paddr=000e2454 vaddr=40088e90 size=0e518h ( 58648) load
I (444) boot: Loaded app from partition at offset 0x10000
I (445) boot: Disabling RNG early entropy source...
I (457) cpu_start: Multicore app
I (465) cpu_start: Pro cpu start user code
I (465) cpu_start: cpu freq: 160000000 Hz
I (465) app_init: Application information:
I (468) app_init: Project name:     zenoh-lib
I (473) app_init: App version:      f61b680-dirty
I (478) app_init: Compile time:     Feb 18 2025 20:46:42
I (484) app_init: ELF file SHA256:  7f1d7e6cb...
I (490) app_init: ESP-IDF:          5.3.1
I (494) efuse_init: Min chip rev:     v0.0
I (499) efuse_init: Max chip rev:     v3.99 
I (504) efuse_init: Chip rev:         v0.0
I (509) heap_init: Initializing. RAM available for dynamic allocation:
I (516) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (522) heap_init: At 3FFB96F0 len 00026910 (154 KiB): DRAM
I (528) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (535) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (541) heap_init: At 400973A8 len 00008C58 (35 KiB): IRAM
I (549) spi_flash: detected chip: gd
I (552) spi_flash: flash io: dio
```

jean-roland, 2025-02-20:
Thanks for the report, will try to reproduce on my end.

jean-roland, 2025-03-20:
Hey @vortex314, tried reproducing on my esp32 without success. However, I do notice that the encoding is consumed by the publication so you need to reapply it on a per packet basis, although it shouldn't cause any crash.

vortex314, 2025-03-26:
ok, will check agaiin