eclipse-zenoh/zenoh-pico #911: [Bug] Write after free in z_msgcodec_test

brandonho667, 2025-03-07:
### Describe the bug

I found an assortment of write after free's in [z_msgcodec_test](https://github.com/eclipse-zenoh/zenoh-pico/blob/main/tests/z_msgcodec_test.c). One example is in [`frame_message`](https://github.com/eclipse-zenoh/zenoh-pico/blob/main/tests/z_msgcodec_test.c), in which `_z_network_message_svec_clear(&msg)` frees `decoded`, which is then written to again 2 lines later (`_z_t_msg_frame_clear(&decoded)`).

For context, I'm currently working on a zenoh-pico interface for [EVL](https://evlproject.org/core/), which has a custom memory allocation implementation that directly writes metadata onto the freed buffer (to organize the heap into an AVL tree). As such, **any** write after free overwrites that metadata and mangles the tree. Could these "write after frees" possibly be eliminated throughout zenoh-pico (perhaps memcheck in CI/CD)?

### To reproduce

1. Build for unix with tests
2. Run `z_msgcodec_test` with `valgrind`
3. See the list invalid reads/writes

### System info

- Platform: Ubuntu 24.04
- CPU: Intel(R) Xeon(R)
- Zenoh version: 7b2b3e606cc2c28b399a40fac1cdba3ab15bbf82

jean-roland, 2025-03-10:
Thanks for the reporting @brandonho667, the most time efficient way to fix this would be to let you push a PR if you feel like doing it.

Otherwise we'll work on it at a later Time™.

brandonho667, 2025-03-11:
gotcha thanks @jean-roland fixed in #912, please take a look!

closed 2025-03-13 by brandonho667