eclipse-zenoh/zenoh-pico #918: [Bug] Combination of features creates compilation issues

vortex314, 2025-03-21:
### Describe the bug

As I port Zenoh Pico to stm32 with stm32cube , I noticed some compilation issues as I reduce the functionality with the feature flags .
I'll update here .
- De-activating Z_FEATURE_MULTICAST_TRANSPORT
```C
Compiling .pio/build/tiva/libb1c/zenoh-pico/transport/raweth/tx.o
lib/zenoh-pico/src/transport/multicast/transport.c:240:6: error: conflicting types for '_z_multicast_transport_clear'
  240 | void _z_multicast_transport_clear(_z_transport_t *zt) { _ZP_UNUSED(zt); }
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from lib/zenoh-pico/src/transport/multicast/transport.c:24:
lib/zenoh-pico/include/zenoh-pico/transport/multicast/transport.h:32:6: note: previous declaration of '_z_multicast_transport_clear' was here
   32 | void _z_multicast_transport_clear(_z_transport_multicast_t *ztm, bool detach_tasks);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/tiva/libb1c/zenoh-pico/transport/transport.o
```
- De-activating Z_FEATURE_INTEREST
```
Compiling .pio/build/tiva/libb1c/zenoh-pico/collections/intmap.o
cc1plus: warning: command line option '-Wno-missing-prototypes' is valid for C/ObjC but not for C++
lib/zenoh-pico/src/api/api.c: In function 'z_declare_publisher':
lib/zenoh-pico/src/api/api.c:1042:58: error: '_z_publisher_t' {aka 'struct _z_publisher_t'} has no member named '_filter'
 1042 |         _z_write_filter_create(_Z_RC_IN_VAL(zs), &int_pub._filter, keyexpr_aliased, _Z_INTEREST_FLAG_SUBSCRIBERS);
      |                                                          ^
lib/zenoh-pico/src/api/api.c: In function 'z_publisher_put':
lib/zenoh-pico/src/api/api.c:1121:41: error: 'z_loaned_publisher_t' {aka 'const struct _z_publisher_t'} has no member named '_filter'
 1121 |         if (!_z_write_filter_active(&pub->_filter)) {
      |                                         ^~
lib/zenoh-pico/src/api/api.c: In function 'z_publisher_get_matching_status':
lib/zenoh-pico/src/api/api.c:1269:42: error: 'z_loaned_publisher_t' {aka 'const struct _z_publisher_t'} has no member named '_filter'
 1269 |     matching_status->matching = publisher->_filter.ctx->state != WRITE_FILTER_ACTIVE;
      |                                          ^~
*** [.pio/build/tiva/libb1c/zenoh-pico/api/api.o] Error 1
```

### To reproduce

Just set certain combinations of feature flags.
As the combination of features can be huge, maybe consider automating the diverse combinations

### System info

- MacOs Or Linux , it's platform independent
- target platform are stm32f401 and lm4f120

vortex314, 2025-03-21:
The use of advanced printf options leads to issues on some platforms , even when the compiler is C++17 
```
    _Z_DEBUG(">>> Allocating interest for (%ju:%.*s)", (uintmax_t)intr->_key._id,
```
Comes out as : 
```
[00:00:15.424 DEBUG ::_z_register_interest] >>> Allocating interest for (ju:� ��5�5�5�5�5�)
```

jean-roland, 2025-03-25:
Thanks for the report on the config token, the first one is already fixed but not the interest one. Not sure what can be fixed for the printf however.

closed 2025-03-25 by sashacmc

vortex314, 2025-03-27:
Maybe , include this one : https://github.com/mpaland/printf or limit the use of advanced features. 
The above version looks very complete on what is used.

jean-roland, 2025-03-27:
We don't want any external dependencies on pico :/

vortex314, 2025-03-27:
It looks to have an MIT license, so I guess it gives you the possibility to copy and merge it. 
But since this is only in ZENOH_DEBUG mode, it's not that important. 
And yeah, C is not Rust, so dependencies can be killing.

PR #921 (2025-03-25): Fix conditional compilation issues
Fixes #918