eclipse-zenoh/zenoh-pico #919: Make zenoh-pico compilation open-ended on platforms

vortex314, 2025-03-21:
### Describe the feature

The zenoh-pico source tree now contains certain combinations of platforms and frameworks that are supported.
When trying to port this to other platforms , I needed to fork the source tree and change some files
Suggestion : if the platform combination is not known or supported just warn this and don't block 
Needed changes : 
- extra_script.py -> should ignore all platform specific source files if unknown ZENOH_GENERIC platform is encountered
- platform.h should give a "#warn" instead of a "#error"
- platform.h should refer to a "zenoh_generic.h" include 
- library.json should list all platforms known to platformio ( maybe there is a way to override this )

Doing this permits to keep system.cpp, network.cpp, zenoh_generic.h at the developer side 
I used this in the fork :  https://github.com/vortex314/zenoh-pico/tree/main

The result  is that a pio project can include zenoh-pico and make the effort of the porting without a need of forks and local copies. The config.h settings remain a challenge however

Example project : https://github.com/vortex314/zenoh-projects/tree/main/zenoh-stm32

Example platformio.ini
```ini
[env:stm32_stm32cube]
platform = ststm32
board = nucleo_f401re
framework = stm32cube
monitor_speed = 115200

lib_deps = 
	lib/zenoh-pico
	lib/stm32_stm32cube

build_flags = 
	-D__GLIBCXX_ASSERTIONS
	-DZENOH_GENERIC
	-DZENOH_DEBUG=3
	-std=gnu++17
	-Wno-error=deprecated-declarations
	-Ilib/stm32_stm32cube
	-g
;	-Os

build_unflags = 
	-std=gnu++11
```

vortex314, 2025-03-21:
See also : #835

vortex314, 2025-03-21:
See also : #818

vortex314, 2025-03-21:
See also : #241

jean-roland, 2025-03-25:
I agree to the need to do something on this front. Feel free to open a PR with your work on this.

vortex314, 2025-04-02:
See pull request. Worked at least for the Texas instruments Tiva lm4f120 and strm32cube. 
Let me know what you think about the approach.

closed 2025-04-07 by gabrik

jean-roland, 2025-04-07:
Fixed with #933