eclipse-zenoh/zenoh-pico #926: [Bug] ze_serializer is quite resource intensive

PetervdPerk-NXP, 2025-04-01:
### Describe the bug

I've got a zenoh-pico node that publishes at a rate of 200Hz on a 160Mhz Cortex-M7 core.
It serializes this msg using CDR with the cyclonedds cdrstream library.
When just using that the CPU load of my zenoh process is around ~2.1%.
When using the following code to add an attachment for rmw_zenoh my cpu load bumps to ~11%.
```
	z_owned_bytes_t attachment;
	z_bytes_empty(&attachment);

	ze_owned_serializer_t serializer;
	ze_serializer_empty(&serializer);

	ze_serializer_serialize_str(z_loan_mut(serializer), "sequence_number");
	ze_serializer_serialize_int64(z_loan_mut(serializer), this->sequence_number++);

	ze_serializer_serialize_str(z_loan_mut(serializer), "source_timestamp");
	ze_serializer_serialize_int64(z_loan_mut(serializer), hrt_absolute_time());

	ze_serializer_serialize_str(z_loan_mut(serializer), "source_gid");
	ze_serializer_serialize_buf(z_loan_mut(serializer), rmw_gid, 16);

	ze_serializer_finish(z_move(serializer), &attachment);
	options.attachment = z_move(attachment);

```

### To reproduce

-

### System info

-

DenisBiryukov91, 2025-04-04:
@PetervdPerk-NXP Do you copy strings like "sequence_number", "source_timestamp", etc. to cdr ?

PetervdPerk-NXP, 2025-04-04:
@DenisBiryukov91 No CDR is a binary format so not needed.

DenisBiryukov91, 2025-04-07:
> [@DenisBiryukov91](https://github.com/DenisBiryukov91) No CDR is a binary format so not needed.

So this is the explanation I believe. Serializing 3 ints should clearly require less resources than serializing 3 ints + 3 variable length strings - which requires at least a couple of heap memory allocations.

I'm not sure I understand, why you need to serialize strings though, you could just serialize only 3 ints and deserialize them in the same order.

DenisBiryukov91, 2025-04-07:
Opened related issue in rmw_zenoh repo https://github.com/ros2/rmw_zenoh/issues/600 which should address performance issues related to attachment structure.

DenisBiryukov91, 2025-04-07:
> Overall I would expect that serializing 71 bytes @ 200hz shouldn't eat 8% CPU load.

As I mentioned it will make at multiple calls to allocate heap memory (I believe a total of 3 for your case), given that total size of data is not known in advance.
The solution might be to allow specifying initial buffer size, to minimize number of allocations.

Do you mind checking performance if you only serialize a single string of 71 bytes in one pass ?

PetervdPerk-NXP, 2025-04-07:
Did some testing with different attachments.

| Method                           | Load |
|----------------------------------|------|
| No attachment                    | 2%   |
| Full serialization               | 11%  |
| Only serialize values            | 5%   |
| Custom stack-based serialization | 3%   |

Full serialization
```
	ze_serializer_serialize_str(z_loan_mut(serializer), "sequence_number");
	ze_serializer_serialize_int64(z_loan_mut(serializer), this->sequence_number++);

	ze_serializer_serialize_str(z_loan_mut(serializer), "source_timestamp");
	ze_serializer_serialize_int64(z_loan_mut(serializer), hrt_absolute_time());

	ze_serializer_serialize_str(z_loan_mut(serializer), "source_gid");
	ze_serializer_serialize_buf(z_loan_mut(serializer), rmw_gid, 16);
```

Only serialize values

```
	ze_serializer_serialize_int64(z_loan_mut(serializer), this->sequence_number++);
	ze_serializer_serialize_int64(z_loan_mut(serializer), hrt_absolute_time());
	ze_serializer_serialize_buf(z_loan_mut(serializer), rmw_gid, 16);
```

Custom stack-based serialization (kinda faulty just quickly based on decoded z_serializer output)
```
	uint8_t attachment_stack[33]; // Somehow ze_serializer goes to 33
	*(int64_t*)&attachment_stack[0] = this->sequence_number;
	*(int64_t*)&attachment_stack[8] = hrt_absolute_time();
	attachment_stack[17] = 0x10; // Magic byte probably wrong but this is just perf test
	memcpy(&attachment_stack[17], rmw_gid, 16);

	z_owned_bytes_t attachment;
	z_bytes_copy_from_buf(&attachment, attachment_stack, size);
	options.attachment = z_move(attachment);
```

JEnoch, 2025-04-08:
ros2/rmw_zenoh#601 that removes the strings from attachments has been merged to `rolling` just before code freeze for Kilted. It also has been backported to `jazzy`.