eclipse-zenoh/zenoh-pico #945: zenoh-pico headers does not compile under c++

alex-botti, 2025-04-18:
### Describe the release item

I'm trying to include zenoh-pico api in my c++ project in visual studio, but i have this kind of error:

C4576	a parenthesized type followed by an initializer list is a non-standard explicit type conversion syntax

for all the inline functions of the api that use of round parenthesis around the returned struct, for example:

static inline _z_slice_t _z_slice_null(void) { return **(**_z_slice_t**)**{0}; }

this is a C11 only construct and not compatible with C++.

So to make it works, function implementations that make use of this sintax should be moved from inline to .c files,

jean-roland, 2025-04-29:
Alternatively, can you deactivate C4576? I'd rather avoid changing this because of how Windows compiler handles c inclusion in c++ and because this has an impact on performance.

alex-botti, 2025-05-22:
Ok i did as you suggested, deactivated C4576 in VS, thanks !

closed 2025-05-22 by alex-botti