eclipse-zenoh/zenoh-pico #956: The zenoh pico does not build for Arduino by platformIO.

hachi-kun, 2025-05-12:
### Describe the bug

I generated a platformio workspace following this README information.

>$ mkdir -p /path/to/project_dir
$ cd /path/to/project_dir
$ platformio init -b az-delivery-devkit-v4
$ platformio run

I try build target image of arduino by platformio.
but, this build is stop.

### To reproduce

 -- build log

<pre>
% pio run
Processing az-delivery-devkit-v4 (platform: espressif32; board: az-delivery-devkit-v4; framework: arduino)
--------------------------------------------------------------------------------------------------------------
Library Manager: Installing FreeRTOS @ 1.0.0
Library Manager: Warning! More than one package has been found by FreeRTOS @ 1.0.0 requirements:
Library Manager:  - feilipu/FreeRTOS@11.1.0-3
Library Manager:  - mbed-rgrover1/FreeRTOS@0.0.0+sha.8e57f3e9cc89
Library Manager:  - mbed-fep/freertos@0.0.0+sha.62cd296ba2a7
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
PLATFORM: Espressif 32 (6.10.0) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting main.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Run command: cmake /home/katsumi/PlatformIO/arduino_zenoh_pico/.pio/libdeps/az-delivery-devkit-v4/zenoh-pico ['-DFRAG_MAX_SIZE=4096', '-DBATCH_UNICAST_SIZE=2048', '-DBATCH_MULTICAST_SIZE=2048']
-- Setting C11 as the C Standard
Compilers in use: GNU,
-- Building with feature config:
* UNSTABLE_API: 0
* MULTI-THREAD: 1
* PUBLICATION: 1
* SUBSCRIPTION: 1
* QUERY: 1
* QUERYABLE: 1
* LIVELINESS: 1
* INTEREST: 1
* AUTO_RECONNECT: 1
* MATCHING: 0
* RAWETH: 0
-- Building in RELEASE mode
-- Build shared library: ON
-- Zenoh Level Log:
-- Fragmented message max size: 4096
-- Unicast batch max size: 2048
-- Multicast batch max size: 2048
-- Build for Zephyr RTOS: OFF
-- Build for FreeRTOS-Plus-TCP: OFF
-- Build for FreeRTOS + LWIP: OFF
-- Build for Raspberry Pi Pico: OFF
-- Configuring for Linux
-- Produce Debian and RPM packages: OFF
-- Build examples: ON
-- Build tools: OFF
-- Build tests: ON
-- Build integration: OFF
-- AddressSanitizer: OFF
-- zenoh-pico examples
-- Configuring done (0.0s)
-- Generating done (0.1s)
-- Build files have been written to: /home/katsumi/PlatformIO/arduino_zenoh_pico/.pio/libdeps/az-delivery-devkit-v4/zenoh-pico/build
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- zenoh-pico @ 1.2.1+0.sha.ede74c3
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/az-delivery-devkit-v4/src/main.ino.cpp.o
Building .pio/build/az-delivery-devkit-v4/bootloader.bin
Generating partitions .pio/build/az-delivery-devkit-v4/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio/build/az-delivery-devkit-v4/lib822/BluetoothSerial/BTAddress.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib822/BluetoothSerial/BTAdvertisedDeviceSet.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib822/BluetoothSerial/BTScanResultsSet.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/lib822/BluetoothSerial/BluetoothSerial.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/api/api.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/api/encoding.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/api/liveliness.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/api/serialization.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/arc_slice.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/bytes.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/fifo.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/fifo_mt.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/intmap.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/lifo.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/list.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/lru_cache.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/refcount.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/ring.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/ring_mt.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/slice.c.o
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/string.c.o
.pio/libdeps/az-delivery-devkit-v4/zenoh-pico/src/collections/refcount.c:122:2: error: #error "Multi-thread refcount in C99 only exists for GCC, use GCC or C11 or deactivate multi-thread"
 #error "Multi-thread refcount in C99 only exists for GCC, use GCC or C11 or deactivate multi-thread"
  ^~~~~
Compiling .pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/vec.c.o
*** [.pio/build/az-delivery-devkit-v4/libba4/zenoh-pico/collections/refcount.c.o] Error 1
========================================= [FAILED] Took 5.36 seconds =========================================
</pre>

### System info

build host : ubuntu24.04
platformio : PlatformIO Core, version 6.1.18

jean-roland, 2025-05-13:
Hello @hachi-kun, this is an issue in our documentation that we need to fix but we had to add compile time definitions in order to build zenoh-pico for a given platform. In your case you need to add the following to your platform.ini (or in the CLI):

```
build_flags=
	-DZENOH_COMPILER_GCC
	-DZENOH_ESPIDF
```

closed 2025-05-13 by jean-roland

hachi-kun, 2025-05-13:
jean-san:

thunks your advice.

I have one question.
Is there any reason not to modify extra_script.py?

> [katsumi@tama](develop_platfromio) % git diff HEAD~
> diff --git a/extra_script.py b/extra_script.py
> index 6032a6b8..c6d019c4 100644
> --- a/extra_script.py
> +++ b/extra_script.py
> @@ -49,7 +49,7 @@ elif FRAMEWORK == 'arduino':
>                        "-<system/flipper/>",
>                        "-<system/windows/>",
>                        "-<system/zephyr/>"]
> -        CPPDEFINES = ["ZENOH_ARDUINO_ESP32", "ZENOH_C_STANDARD=99"]
> +        CPPDEFINES = ["ZENOH_ARDUINO_ESP32", "ZENOH_C_STANDARD=99", "ZENOH_COMPILER_GCC"]
>      if PLATFORM == 'ststm32':
>          BOARD = env.get("PIOENV")
>          if BOARD == 'opencr':

jean-roland, 2025-05-13:
Well I didn't think of this, if it works I'd suggest you to open a PR for that please.

hachi-kun, 2025-05-13:
I see.

If you can confirm that zenoh-pico works by just adding this compile option(ZENOH_COMPILER_GCC,ZENOH_ESPIDF), then there is no problem.

jean-roland, 2025-05-13:
Yes, it should work with those compile options.