eclipse-zenoh/zenoh-pico #982: [Bug] Wrong session open procedure when scouting

spynetcamera, 2025-07-06:
### Describe the bug

In `src\net\session.c`, `_z_open()` performs scouting when no locator is provided, so that remote locators can be discovered and connected.
But `peer_op` defaults to `_Z_PEER_OP_LISTEN` and, when a peer with no locator is used, this would lead in using remote endpoints (from scouting) as they where local endpoints.
`_z_open_inner(..., peer_op)` will call `_z_new_transport_peer(..., peer_op)` and finally `_z_listen_link()`. This will fail while trying to bind a remote address.

### To reproduce

N/A

### System info

Tested in v1.4.0.

jean-roland, 2025-07-07:
Hello @spynetcamera could you give me more details on your setup, what kind of topology / links are you trying to establish?

jean-roland, 2025-07-07:
I mean I can fix the issue, but peer unicast connections to scouted endpoints might be problematic since pico does not negotiate connection parameters, so you'll need to have same config (especially in buffer sizes) on both ends.

closed 2025-07-07 by OlivierHecart

spynetcamera, 2025-07-11:
> completed via #983 <https://github.com/eclipse-zenoh/zenoh-pico/pull/983>.

Hello,

Thank you for the fix and sorry for my late reply!
I'm going to port Zenoh-pico to Android and try to set it as a peer, so
that it can participate in the network even if other peers or routers come
later.
Honestly I'm a little confused about interaction with Zenoh and Zenoh-pico,
what are the roles of peers and clients in this configuration and such.
Thank you for sharing some documentation if any.
Regards!
Spynet

jean-roland, 2025-07-12:
First, why don't you use Zenoh Kotlin to interface on Android?

Peer or client just change behavior a bit. Client will negotiate buffer sizes with router, peer will not so you need to match those for connection to be accepted.

Client will only have one connection to a router, peer may accept n connections in unicast

PR #983 (2025-07-07): Fix scout peer op
Fixes #982