eclipse-zenoh/zenoh-pico #1010: Unicast Read Thread: A Proposed Solution for Excessive CPU Usage and a Question on Connection Termination Detection

42dot-bongjun-hur, 2025-08-06:
### Describe the release item

In Zenoh-Pico's unicast transport, the read thread consumes excessive CPU resources when there is no data to receive. This occurs because the thread is busy-waiting on a non-blocking socket, immediately retrying the read operation in a tight loop. To fix this, I propose adding a short sleep (z_sleep_ms(1)) after an unsuccessful read to allow the CPU to yield to other tasks. This simple change will significantly reduce CPU usage, which is critical for resource-constrained embedded systems.

zenoh-pico\src\transport\unicast\read.c line 358:
```
            size_t to_read = 0;
            // Retrieve data
            if (!_z_unicast_client_read(ztu, curr_peer, &to_read)) {
                // [Proposed Change] Add a small delay to prevent busy-waiting
                z_sleep_s(1);
                continue;
            }
```

**Additionally, I would like to know how a Zenoh-pico client can detect when its TCP connection to a broker has been terminated. What is the proper way to handle this state in the client application?**

jean-roland, 2025-08-07:
Hello @42dot-bongjun-hur, the read thread was designed to use blocking sockets, adding a sleep would introduce a latency as high as the sleep so it's not an acceptable solution.

What platform are you using? If for some reason you are forced to use non-blocking socket, you can try setting your zenoh-pico nodes as unicast peer as they handle non-blocking sockets.

For lost connection, zenoh-pico has the ability to try to reconnect (`Z_FEATURE_AUTO_RECONNECT`).

42dot-bongjun-hur, 2025-08-08:
Hi @jean-roland, Thank you for the clarification. I understand the issue with adding sleep(), and your explanation makes sense.
Regarding reconnection, since the design assumes blocking sockets by default, I also think it is appropriate for the _zp_unicast_read_task() operation to detect when the broker has been shut down and return an error in that case. From my perspective, the AUTO_RECONNECT feature you mentioned seems like a broader recovery solution. My question is: when the TCP connection is lost, does the read task also return an error, or is it handled differently as part of the auto-reconnect process?